StuckAtPrototype/Racer #1: PLZ Add PDF for Schematic and PCB artwork

puviarasud, 2024-10-17:
Hi good project. Please add a PDF copy of the schematic and PCB artwork for people to have a look at your design. not all have the tools installed . Thanks

oscarpeters, 2024-10-17:
Been working on improving readability of the schematics.
[RACER-v0.2.1.pdf](https://github.com/user-attachments/files/17410684/RACER-v0.2.1.pdf)

puviarasud, 2024-10-18:
Thanks , put it in the respective Github folder. I will look into it and suggest improvements, All the best.

StuckAtPrototype, 2024-10-19:
> Been working on improving readability of the schematics. [RACER-v0.2.1.pdf](https://github.com/user-attachments/files/17410684/RACER-v0.2.1.pdf)

if you want to open up a pull request I am happy to merge it in :)