StuckAtPrototype/Racer #3: Suggestion:  Add FPV capability via using esp32 cam.

villimikko, 2024-10-17:
Great project! Look really fun!

I think possibly adding FPV would make it perfect ( not sure if controller range is a limit)
Here is example project that is showing it is possible.

https://github.com/RomanLut/hx-esp32-cam-fpv

StuckAtPrototype, 2024-10-18:
neat! this would be fun for sure. I fly FPV freestyle quads and love FPV in general. I might make some youtube shorts showcasing the car with a chepo tiny whoop camera on it :)

AirborneEagle, 2024-10-18:
I would certainly be there for that!
To I used to be quite deep into the fpv drone racing world. 
I've also built custom fpv RC cars for a small business that I started once. 
Thank you for sharing this awesome project!

StuckAtPrototype, 2024-10-19:
I have analog FPV cameras, I used to be deep into freestyle FPV. I even designed my own frames! I still love my frame more than any other out there, sadly not much time to fly these days. I might do a completely from scratch hardware and firmware as a project one day. I know there is a ton of open source stuff, but I have some fun twists in mind :) 

That said, I might do a short running around the house with an FPV camera on one of these!