StuckAtPrototype/Racer #4: Please add BOMs

alexspurling, 2024-10-17:
It would be helpful to include BOM files for the PCB (including the motors) so we know which parts to order (same goes for the controller project).

tobiasoort, 2024-10-17:
(this way more novice users can use a service like JLCPCB to order the completed PCB and get going quickly too)

Rubines, 2024-10-17:
Thank you for sharing this fun project!
Please add the motor data, do you have a link where I could source them ?

mianzhi, 2024-10-17:
> Thank you for sharing this fun project! Please add the motor data, do you have a link where I could source them ?

I guess they are "6 mm geared coreless motor, 1200 rpm nominal speed at 3V". @StuckAtPrototype thanks for sharing the project, please correct me if i was wrong.

StuckAtPrototype, 2024-10-17:
@mianzhi thats correct!

I will be sharing a BOM late tonight!

for now, you can find the motors by searching "6mm planetary motors 1200 rpm"

Rubines, 2024-10-17:
Thank you both for your help!
@mianzhi @StuckAtPrototype

StuckAtPrototype, 2024-10-18:
initial BOM added for the racer. I still have the sensor board and the controller to do, but that will have to wait until tomorrow 👍

hazerty565, 2024-10-18:
hi, I would like to know whats is motor it is use? have you got a reference? Thank you

StuckAtPrototype, 2024-10-19:
All BOMs are in! if there are any problems, lets open a new issue. closing this out 🎉

closed 2024-10-19 by StuckAtPrototype