StuckAtPrototype/Racer #8: Use USB Pins

majuss, 2024-10-17:
Hi there,

thank you for sharing your project!  The H2-mini is a "C6 variant" which means it comes with onboard USB, no programmer etc. neccessary! Why didn't you connect the D- and D+ lines of the USB-C port to the MCU. That way you could easily program it via USB. That way you can skip the UART header entirely.

StuckAtPrototype, 2024-10-19:
Hi! thats so true! I will do that for the next project. Thanks!! For this project, I would have to upgrade the connector as the one I used does only power. But this is a very good idea. Ill keep it in mind for the project I am working on now :) Closing this out for this one. thanks again!

andistorm, 2024-11-06:
Nice idea,

currently there is only a 6-pin USB-C connector used. If i am not wrong, you would need at least 8 pins, with `D-` and `D+`.

This would mean we need to change pcb, which we try not to do for `v0.2.0` release, but after checking at JLCPCB I realized this connector is currently not available, so maybe we need to change USB-C Connector anyways.

Since this issue is in discussion again, I reopen it.

What do you think @StuckAtPrototype