StuckAtPrototype/Racer #13: oshwa

mbz4, 2024-10-18:
please consider certifying this project with oshwa: https://certification.oshwa.org/

may help sort out licensing and help open the doors to more public dev engagement

JeremyDWilliams, 2024-10-19:
It's already licensed,
maybe you missed that :)

andistorm, 2024-11-06:
Hey @mbz4,

thanks for pointing this out. I agree, having this project certified as "Open Source Hardware matching community definition" would be quite nice. To do this we would need someone who collects points that needs to be adapted. 

If you’d like to move this forward, please feel free to post a list here in the issue with the items that would need to be changed. Additionally, a summary of how the certification process would work would also be helpful.

Cheers 🐧