StuckAtPrototype/Racer #14: CPL file

AirborneEagle, 2024-10-18:
I am totally new to the world of PCB making and assembly. 
Do you have a cpl file that you could include?

StuckAtPrototype, 2024-10-18:
ill add it to the list, thats a good point!

TexGG, 2024-10-19:
I was gonna ask the same. The project look amasing, the CPL will hepl produce the circuit easier :)

mr-ukas, 2024-10-30:
I created a pick&place file from Kicad. First time in my life, so better double check : D
For getting a PCBA-manufacturer's quote, it should be good enough 👍 

PCB top edited to match JLCPCB requirements to format. 
[racer-top-pos.csv](https://github.com/user-attachments/files/17579037/racer-top-pos.csv)
[CPL_miniRacer_position_pickNplace_JLC_top.xlsx](https://github.com/user-attachments/files/17579038/CPL_miniRacer_position_pickNplace_JLC_top.xlsx)
[racer-bottom-pos.csv](https://github.com/user-attachments/files/17579039/racer-bottom-pos.csv)

mr-ukas, 2024-10-30:
all revision V0.2

mr-ukas, 2024-10-30:
`What do you mean by this? Is it not based on master branch?`
It's the master branch. I just wanted to add this information about the revision, in case somebody will update the revision in the future. 

`where you explain the steps you have done 👀`

I just re-named the columns according to this guide here:
https://jlcpcb.com/help/article/pick-place-file-for-pcb-assembly

andistorm, 2024-10-30:
>I just re-named the columns according to this guide here

Oh wow, you typed in all the rows by hand? I would have thought there is some tooling to  achieve this.

Did you have a look at [this guide](https://jlcpcb.com/help/article/How-to-generate-the-BOM-and-Centroid-file-from-KiCAD)?

mr-ukas, 2024-10-30:
> Did you have a look at [this guide](https://jlcpcb.com/help/article/How-to-generate-the-BOM-and-Centroid-file-from-KiCAD)?

yes, true, this page was my starting point. 
I used also the MS excel to import the CSV-file to a regular excel sheet. This transferred the rows to proper columns, which had to be re-named ✌️

mr-ukas, 2024-10-30:
Feel free to creat a pull request to merge the files. I have no idea how this is done. It's late here :D

andistorm, 2024-10-30:
> Feel free to creat a pull request to merge the files.

Probably I will, but I want to try to generate this files by myself for this

fschrempf, 2024-11-07:
I'm usually using [Aisler](https://aisler.net/) to order PCBs (and optionally parts/assembly). They allow to upload the KiCad PCB file and get all the required data from it. For matching the components automatically it requires a ["MPN" field](https://community.aisler.net/t/documenting-parts/56) to be set as the "Manufacturer Part Number". I don't know the requirements for JLCPCB, but it would be great if we could also add the MPN field.