StuckAtPrototype/Racer #26: sdkconfig missing

paolosanchi, 2024-10-29:
Hello, I'm trying to compile and run the firmware on an esp32 wroom devkit, I currently don't have a esp mini 1, but I guess it should be compatible.
I could compile it by changing the default configuration (I basically had to activate the BLE).
The file sdkconfig is missing and I'm afraid I miss some kind of configurations.
The firmware runs but it seems to get stuck.
this is the output:

```
I (31) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (31) boot: compile time Oct 29 2024 05:32:28
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1a2f8h (107256) map
I (141) esp_image: segment 1: paddr=0002a320 vaddr=3ff80000 size=00004h (     4) load
I (141) esp_image: segment 2: paddr=0002a32c vaddr=3ffbdb60 size=04dech ( 19948) load
I (154) esp_image: segment 3: paddr=0002f120 vaddr=40080000 size=00ef8h (  3832) load
I (157) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=595c0h (366016) map
I (289) esp_image: segment 5: paddr=000895e8 vaddr=40080ef8 size=18570h ( 99696) load
I (341) boot: Loaded app from partition at offset 0x10000
I (341) boot: Disabling RNG early entropy source...
I (353) cpu_start: Multicore app
I (362) cpu_start: Pro cpu start user code
I (362) cpu_start: cpu freq: 160000000 Hz
I (362) app_init: Application information:
I (365) app_init: Project name:     hello_world
I (370) app_init: App version:      1807a4c-dirty
I (375) app_init: Compile time:     Oct 29 2024 12:30:22
I (381) app_init: ELF file SHA256:  8ac5c9a1f...
I (387) app_init: ESP-IDF:          v5.3.1
I (391) efuse_init: Min chip rev:     v0.0
I (396) efuse_init: Max chip rev:     v3.99
I (401) efuse_init: Chip rev:         v3.0
I (406) heap_init: Initializing. RAM available for dynamic allocation:
I (413) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (419) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (425) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (431) heap_init: At 3FFC5CC8 len 0001A338 (104 KiB): DRAM
I (438) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (444) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (450) heap_init: At 40099468 len 00006B98 (26 KiB): IRAM
I (458) spi_flash: detected chip: winbond
I (461) spi_flash: flash io: dio
W (465) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
W (479) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (490) coexist: coex firmware version: 4482466
I (495) main_task: Started on CPU0
I (505) main_task: Calling app_main()
I (505) gpio: GPIO[27]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (505) gpio: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (515) gpio: GPIO[25]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (525) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
LED flash period set to 10 ticks
LED 0 (GPIO 27) set to ON
LED 1 (GPIO 26) set to ON
LED 2 (GPIO 25) set to ON
LED 3 (GPIO 22) set to ON
LED 3 (GPIO 22) set to OFF
LED 2 (GPIO 25) set to OFF
LED 1 (GPIO 26) set to ON
LED 0 (GPIO 27) set to ON
I (565) BTDM_INIT: BT controller compile version [b022216]
I (565) BTDM_INIT: Bluetooth MAC: 40:22:d8:57:e3:b6
I (575) phy_init: phy_version 4830,54550f7,Jun 20 2024,14:22:08
I (895) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| P
```
As you can see the last line is interrupted

StuckAtPrototype, 2024-10-30:
Hi, Ill add the sdkconfig, thats a good point. I never ran this firmware on anything but the esp32-h2, so not sure it would run. there could be a GPIO collision from what it looks like

paolosanchi, 2024-10-31:
Ok, I wait for the sdkconfig, meanwhile I'll get an esp-h2 devboard and try to build my own car.
I really love this project and I'd like to help in its development.
I've texted you in private in instagram, please check out your message requests.

StuckAtPrototype, 2024-11-03:
sdkconfig added

closed 2024-11-03 by StuckAtPrototype