StuckAtPrototype/Racer #28: Missing footprint for USB C Connector

andistorm, 2024-10-30:
## My Setup

kicad 8.0.6

## How To Reproduce

1. Checkout `master`branch
2. Open PCB
3. Update PCB from Schematic

## Error Log

```
Error: P1 pad A4 not found in kicad_reformed_custom_IC:CUI_UJC-VP-3-SMT-TR.
Error: P1 pad B4 not found in kicad_reformed_custom_IC:CUI_UJC-VP-3-SMT-TR.
Error: P1 pad A1 not found in kicad_reformed_custom_IC:CUI_UJC-VP-3-SMT-TR.
Error: P1 pad B1 not found in kicad_reformed_custom_IC:CUI_UJC-VP-3-SMT-TR.
Error: P1 pad A6 not found in kicad_reformed_custom_IC:CUI_UJC-VP-3-SMT-TR.
Error: P1 pad A7 not found in kicad_reformed_custom_IC:CUI_UJC-VP-3-SMT-TR.
```

## Explanation

Opening the schematic file show that `kicad_reformed_custom_IC:CUI_UJC-VP-3-SMT-TR` is used as the footprint.
The library `kicad_reformed_custom_IC` comes not with default kicad installation. I guess it was imported once, after choosing the USB C Connector.
The problem is one could only guess which USB C Connector should be used.

## Open Questions

1. Which USB C connector is used?
2. Should we add a custom footprint library to the project? (including this one)

StuckAtPrototype, 2024-11-03:
Thanks for raising the issue, I will create a separate repo with all my footprints, thats should help in the future as well. Connector is CUI_UJC-VP-3-SMT-TR

StuckAtPrototype, 2024-11-03:
added necessary libraries here https://github.com/StuckAtPrototype/StuckAtPrototypeKiCadLibs/tree/master

closing this out for now, though please let me know if it works after setting them up.

andistorm, 2024-11-06:
Unfortunately, that didn’t solve the problem, but I found out that the symbol and footprint didn’t match

See my PR: https://github.com/StuckAtPrototype/Racer/pull/33